TheKikGen/USBMidiKliK4x4 #4: Possible ProcessSysExInternal() filtermask bug?

orzdk, 2019-09-15:
Had some trouble setting filters with sysex, but only on source type 1 serial, so I examined the parsing. Could there be an error in line 738 in UsbMidiKliK4x4.ino:

*filterRouting = (*filterRouting & 0xF0 ) | ( filtersMsk << 4) ;

should be

*filterRouting = (*filterRouting & 0xF0 ) | filtersMsk;

BR Jesper

TheKikGen, 2019-09-16:
A filter is stored in one byte to the EEPROM as follow :
Bits 0:3 = Filter mask for Serial
Bits 4:7 = Filter mask for Cable (USB) 

SO.. YES ! You are totally right Jesper.
Damn copy and paste.  
Since I implemented the configuration via a terminal on the serial port, 
I use less SYSEX....I push a correction in the repository.

TheKikGen, 2019-09-16:
Correction committed to master branch.

closed 2019-09-16 by TheKikGen