TheKikGen/USBMidiKliK4x4 #6: Just a little help ?

AurelGTS, 2019-10-02:
Hello,
This issue is just to start conversation, this is not an issue on your project ...
I m  working on a midi device with a blue pill. I m working with STM32duino USBComposite library.
Your library seems close to this USBComposite library. So I would appreciate your help if possible :) I think it could be quickly fixed.
I have a problem : I can receive data but I can't transmit... If you have a little bit time to help me, I ll be thankful. I think its perhaps : 
- usb descriptors which are not correct.
- bootloader issue but I m not sure...

Thanks in advance.
Regards
Aurélien

TheKikGen, 2019-10-02:
Hello Aurelien,
Have a look in the descriptor here . The STM32DUINO bootloader is not the problem.  Know that the USB composite is sometimes unstable....Start with a basic midi out.

closed 2019-10-02 by TheKikGen

AurelGTS, 2019-10-03:
Hello, 

Thanks for your feedback. Midiout.ino example just work fine. 
I copied all your code and I ve same behavior : no data transmitted. 

I really don't understand. 

regards, 

Aurélien 

----- Mail original -----

TheKikGen, 2019-10-03:
Aurélien, please stop to post here as it is not a general purpose forum.  Post to the Arduino forum, open a new topic with your issue.  Thanks !