TheKikGen/USBMidiKliK4x4 #10: Allow for "blind" serial target configuration ?

orzdk, 2019-11-01:
Hey : )

Setting non-existant serial targets	by sysex config is blocked by line 1045 and 1052. 		
1045  if (dstType  == 1 &&  msgLen > (SERIAL_INTERFACE_COUNT + 5) )  break;
1052  (dstType == 1 && b < SERIAL_INTERFACE_COUNT) ) {

I propose allowing setting blind serial targets, as it will allow for configuration of a bus environment without having all units connected, and should be safe from a routing perspective as the serialOutTargets iterator only counts to SERIAL_INTERFACE_COUNT anyway.

Fixed in my local branch by defining SERIAL_INTERFACE_CONFIG_MAX and using that in 1045,1052, so just a suggestion. 

BR

orzdk, 2019-11-03:
I was too quick to say its solved with the "hack", it seems to be a bit deeper issue. I will investigate more.

TheKikGen, 2019-11-05:
The STM32F103 has only 3 serials ports, it is why the limit is 5 boards in bus mode (5x3 = 15 serial ports) as the midi standard allows 16 cables I/O max.   I don't see why you need to address a non existing serial port . If it is for "virtual routing" purpose,  you could combine routing rules at the cable level, because this is already fully virtual. You can route any of the 16 cables out to any of the 15 physical serial ports or to the 16 cables IN. And you can chain cables routing rules....

orzdk, 2019-11-05:
With regards to a completely assembled solution with a fixed number of STM's and connectors, you are of course right, - then it does not make any sense to allow routing to non-existant target.

I was just thinking that it in some situations could be useful (and harmless) to allow sysex commands to configure slave targets even if the slave is not connected at the time, but it's probably only useful for work in progress integration design situations, so I'll just keep it in my fork. 

BR : )

closed 2019-11-05 by orzdk