TheKikGen/USBMidiKliK4x4 #15: -03 opt fails upload and trashes bootloader

orzdk, 2019-11-13:
This is probably a problem in my environment, but when compiling with -O3 fastest optimization, the upload fails and bootloader gets overwritten and needs to be reloaded.  Any idea why O3 code should trash the bootloader, and what could be done? Tried before and after 4dbdc59, no difference. Thanks for any advice ! : ) BR

Opening USB Device 0x1eaf:0x0003...
Found Runtime: [0x1eaf:0x0003] devnum=1, cfg=0, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
Setting Configuration 1...
Claiming USB DFU Interface...
Setting Alternate Setting ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
Transfer Size = 0x0400
bytes_per_hash=1192
dfu_download error -5
Error during download
Starting download: [################################################
timeout waiting for COM3 serial

TheKikGen, 2019-11-13:
O3 produces huge bin due to optimization done by the compiler.
What is the size of the bin after compilation?
Try O2 also...
Strange that the bootloader overwrites its own flash mem area...

TheKikGen, 2019-11-13:
Try to uncheck agressive caching in the Arduino IDE and compile again...

orzdk, 2019-11-14:
Thank you very much, I will try disabling aggressive caching later today and see if it helps : )

orzdk, 2019-11-14:
All test done on official master, - bin sizes are O3 59,632 bytes and O2 49,144 bytes. 

All optimizations below O3 works, but as you mention, that would almost with certainty be attributed to the bin size. Could be interesting to try to link some big dummy .h file to see if O2 can also be made to err in similar fashion. 

The caching options seems to have been removed again from the IDE, but i tried compiler.cache_core=false and cache.enable=false in preferences.txt, but still happens.  

I don't think this has anything to do with UsbMidiKliK as such, probably a bug in bootloader, board , platform or chinese clone hardware, some of my stash is probably GDs. 

While probably still a bit out of my legue, nevertheless quite interesting issue. I will investigate. Close issue unless you want to pursue yourself, I will comment in case of any findings.

orzdk, 2019-11-14:
Just for fun and to preempt any GD flavored solutions, i tried both gd32f1_frankenmaple.bin and gd32f1_generic_boot_pc13.bin, but neither seem to work at all on my boards. With regards to the working bootloader - this is what first addresses of generic_boot_pc13.bin looks like in device memory after failed upload of sketch in Arduino IDE... Not 100% sure what I am looking at but ST link util compare seems to indicate some corruption. Last diff at 0x080003F0.

[image]

TheKikGen, 2019-11-22:
All my STM32F103C8 board have 128 KB of flash memory in fact.
You can try to erase all pages in STM32 STLINK utility : you will usually see that number of pages is beyond 64K..., reload the bootloader, and test again...

orzdk, 2019-11-22:
Almost forgot about this one. I think i have some 128kb boards in the stash, will try asap. For the 64K board full chip erase did not help, -O3 still stalls and overwrites bootloader, even at just 92-95% utilization. Thanks for comments.

TheKikGen, 2020-03-18:
I investigated on that recently.    

If you use EEPROM emulation to store permanent settings (as the USBMidiKlik4x4 project does), you must take care to set the EEPROM base address that must be beyond the last page occupied by the firmware. in the flash memory.
   
When you firmware is becoming fat, you have the risk to write in the EEPROM area and vice versa if you save some parameters to the flash memory (overlapping effect!).

A page for the STM32F103C is 1K.
The size of the flash memory is 64K , so 64 pages.
The flash memory start 0x08000000 (page0). 
page 1 @0x08000400,
page 2 @0x08000800
.....
page 62 @0x0800F800,.
page 63 @0x0800FC00 (the last one).

So you can define BASE0 to 0x0800F800, and base 1 to 0x0800FC00 (EEPROM object needs 2 pages due to the writes optimization algorithm).
Check that the firmware size does not become too large in order to avoid overlaps.

TheKikGen, 2020-03-19:
To reduce the code size, and keep a good level of performances, you can use the O2 flag.  O3 generates a fat binary.  

You also choose to keep O3 is you compile with the STM32F103CB variant having 128K of flash memory.  Most of Bluepill on the market support 128K of flash memory even if they are sold for 64K.  It is the case for all the Bluepill boards I have.

orzdk, 2020-03-20:
Thank you for the good advice, I will try to set those base adresses and play around with the flags to find the best compromise. I have indeed experienced the overlapping effect a few places when saving big transformer config data, so this may solve many problems. I also have a untested branch somewhere with all intellithru functionality removed which I do not use at the moment, not sure yet how much it will release. 

I have had to abandon all MIDI and electronics work for months due to dayjob and currently have no development setup, but I plan to rebuild my 9x9 soon and get back to coding and test it out and release and document the transformer branch and remote control, - cannot leave so much work unfinished : )

Good to see you back - always looking forward to new releases from the kikgenlabs : )
Thanks, Cheers !

TheKikGen, 2020-03-20:
You can have a look at the MIDITRANS BRANCH. It is still quite experimental however.

closed 2020-03-26 by TheKikGen