TheKikGen/USBMidiKliK4x4 #21: MSGFLTR Double Status Filter not possible to set

orzdk, 2020-05-24:
Sorry if I misunderstand, but MSGFLTR double status option par3 & par4 seems to expect values from 80 to FF which will not go into a sysex message. Parameter range for par3/4 should probably be 0x8-0xE (or +0xF if common/realtime can also be removed) BR

TheKikGen, 2020-06-04:
You are totally right.  This was tested internally not with full sysex...
Now the following table will be used for midi status ids :
```
--------------------------------
|    Midi status ids table     |
|------------------------------|
| noteOffStatus         | 0X08 |
| noteOnStatus          | 0X09 |
| polyKeyPressureStatus | 0X0A |
| controlChangeStatus   | 0X0B |
| programChangeStatus   | 0X0C |
| channelPressureStatus | 0X0D |
| pitchBendStatus       | 0X0E |
| midiTimeCodeStatus    | 0X11 |
| songPosPointerStatus  | 0X12 |
| songSelectStatus      | 0X13 |
| tuneRequestStatus     | 0X16 |
| timingClockStatus     | 0X18 |
| startStatus           | 0X1A |
| continueStatus        | 0X1B |
| stopStatus            | 0X1C |
| activeSensingStatus   | 0X1E |
| systemResetStatus     | 0X1F |
--------------------------------
``` 
I will publish the patch tomorrow after deeper tests.

TheKikGen, 2020-06-04:
```
  // Midi Status double filter
  else  if ( pipe->par1 == 1  ) {
    uint8_t midiStatus = ( pk->packet[1] >= 0xF0 ? pk->packet[1] - 0xE0 : pk->packet[1]>>4 ) ;
    if ( midiStatus == pipe->par3 || midiStatus == pipe->par4  )
        return (pipe->par2 == 0 ? true : false ); //Keep or drop...
    else return true;
  }
```

orzdk, 2020-06-05:
Cool, thanks : ) I've updated the object model in the UI accordingly and will commit after test. 

Another thing related to the filter, - I am thinking if an Exclusive bit-mask filter could also be useful in addition to the Inclusive? So that the MSGFLTR would be as below. 

```
-----------------------------------------------------------------------------------------------
| PipeID  |        par1        |        par2        |        par3        |         par4       |
|---------+--------------------+--------------------+--------------------+--------------------|
| MSGFLTR |     inclusive      | bits mask.         |     00 (unused)    |     00 (unused)    |
|   00    |     filter:0       |  ch voice:0001 (1) |                    |                    |
|         |                    |  Sys.cmn :0010 (2) |                    |                    |
|         |                    |  realtime:0100 (4) |                    |                    |
|         |                    |  sysex   :1000 (8) |                    |                    |
|	  |		       |                    |                    |                    |
|         |     Exclusive      | bits mask.         |     00 (unused)    |     00 (unused)    |
|         |     filter:1       |  ch voice:0001 (1) |                    |                    |
|         |                    |  Sys.cmn :0010 (2) |                    |                    |
|         |                    |  realtime:0100 (4) |                    |                    |
|         |                    |  sysex   :1000 (8) |                    |                    |
|	  |		       |                    |                    |                    |
|         |   MidiStatus       |   include:0        | status1:08-1F      OR  status2:08-1F    |
|         |  double filter:2   |   exclude:1        | (sysex not allowed)|  if par4 unused:0  |
|---------+--------------------+--------------------+--------------------+--------------------|
```

TheKikGen, 2020-06-06:
Sure Jesper.  trivial so accepted !

orzdk, 2020-06-08:
Thanks for including in build ! I will close issue.

closed 2020-06-08 by orzdk