TheKikGen/USBMidiKliK4x4 #23: Cannot set Merge command w/ filter in config menu?

nathangroth, 2020-06-02:
I have a 4x4 board and I can't see how to filter certain merged messages in the Config Menu.
I'm trying to send all MIDI messages (voice, clock & CC) from input jack 1 to output jack 1, but I want to merge ONLY the clock data from input/output 1 with the voice & cc data of jacks 2 & 3.
I can see how to filter the data, and I can see how to set up so that the data being sent out of one output jack can be looped back to come in to other inputs, but I can't see how to filter it in-between the first output and the second input I want certain data to be merged with

TheKikGen, 2020-06-02:
Hi Nathan. Do you have installed the firmware V2.5 ?

TheKikGen, 2020-06-02:
This should work (but not in standalone) : 
```
Route IN1 to jack OUT1 and  Cable IN8 (8 is an  example). 
Filter only clock for IN8,  and merge it with IN 2,3 to target jack 2,3.
```
Use the cable 8 as a kind of virtual port.

With the last 2.5 version, you can do that easier and also in standalone mode.

nathangroth, 2020-06-03:
I just upgraded to 2.5, build 2.200602.1615 and I'm not not seeing any options for filtering in the Config menu at all. 
Beyond that, to see if I have the routing correct this correct, as there are similarities in terminology, If I want to have the setup described above, I need to take JACK IN 1, and send it to JACK OUT 1 and CABLE IN 4. Then I need to make sure CABLE 4 is enabled in the routing for JACK IN 2 & JACK IN 3, which I'm merging and sending to both JACK OUT 2 & JACK OUT 3. 
Now, where can I filter and merge these messages and will this work in standalone mode as I pan to use this without a computer. Thanks for your help, I'll post this in the WIKI if it's better

TheKikGen, 2020-06-03:
From the 2.5 version , filters were removed and are now managed as pipe.
You have 8 virtuals ports you can use  to split or merge filtered real ports. 
So, you want to route :
Jack IN1 -> Jack OUT1
jack IN1 midi clock -> Jacks OUT2, OUT3
Jacks IN2 -> Jack OUT2
Jacks IN3 -> Jack OUT3
right ?
We need to define a pipeline that will use the LOOPBCK pipe to route IN1 clock to the virtual IN1.
Then we'll route the virtual IN1 to Jk OUT2,OUT3.
As all pipelines options  are not yet available in the configuration menu, we'll use SYSEX.

1/ Define the routing for virtual IN1 (type 2, port 00)  -> jk OUT2 OUT3 ( jk=type 1, port list =  01,02)
F0 77 77 78 0F 01 <in port type> <in port> <out port type>[out ports list: nn...nn] F7
`F0 77 77 78 0F 01    02 00  01 01 02 F7`

2/ We'll use the pipeline #1 to send jk IN1 clock msg to virtual IN1.  To do that, we need to add the pipe #6 LOOPBCK to the pipeline 1, with a filter set to midi clock, and a route to virtual 1 and then attach the pipeline 1 to the jk IN1 (Hope I'm clear !)

. Add pipe id 6 to slot 01, par1=port type 2, par2 = filter mask 4, par3= port 00, par4 not used (00) :
F0 77 77 78 11 01 00 <slot:1-8> <pipe id: nn> <prm: nn nn nn nn > F7
`F0 77 77 78 11 01 00    01 06 02 04 00 00  F7`

. Attach slot 1 (pipeline 1) to Jk IN1 :
F0 77 77 78 11 00 02   < in port type> < in port: 0-F> <slot: 0-8> F7
`F0 77 77 78 11 00 02   01 00  01 F7`

closed 2020-07-09 by TheKikGen