TheKikGen/USBMidiKliK4x4 #27: flashing of prebuild binaries?

knopserl, 2020-07-12:
any hint how to flash the 2.5.1 binary(ies)? with an ST-Link flasher?
Do I need the bootloader or how to combine the generic-bootloader + the MIDI firmware (addresses?)
The Bluepill has 128kB flash

orzdk, 2020-07-12:
I have never tried the prebuilt binaries or the provided flash tools, but you can get it to work by first burning the generic_boot20_pc13.bin from the bin folder of this repository, using STLink utility. Then disconnect the STLink and connect USB, open the project .INO and select the "STM32duino bootloader" upload method and build. Sometimes needed to click reset button at moment of upload.

knopserl, 2020-07-13:
Thanks, that is what I finally did and what worked (took me also some time). I just wanted to know what addresses I need if I flash both, the bootloader and the MIDI part. I have not found hints for flashing both binaries. Obviously you need a different address for the MIDI firmware part.

TheKikGen, 2020-07-14:
You can flash without any bootloader (only st-link) at 0x08000000 but you must recompile and use "serial" in the upload options in the Arduino IDE.   

Otherwise, download the last 2.51 release here : 
 https://github.com/TheKikGen/USBMidiKliK4x4/releases/tag/v2.5.1 , 

Then upload the tkg-hid-bootloader firmware here : 
https://github.com/TheKikGen/stm32-tkg-hid-bootloader/blob/master/F1/bootloader_only_binaries/tkg_hid_generic_pc13.bin

and flash it with st-link  at 0x08000000. This new bootloader doesn"t need any driver.

Then use TKG-FLASH.EXE provided in the 2.51 release  to flash the 128K or the 64k bluepill version.  
The 128K firmware should work on a 64K Bluepill. if it is not the case, try the 64K one.

knopserl, 2020-07-17:
Ok, thanks, I managed to do the upload by using the generic STM bootloader and used the the Arduino IDE and do the updload by compile/upload with the bootloader als upload in the IDE. But I'll anyhow do another flash using your latest hints.

jalkjaer, 2020-09-17:
You can simply put the firmware after the bootloader (0x1000 = 4096 which is the size of the bootloader)
`st-flash --flash=128k umk4x4.stm32f103cb-128k-o3.bin 0x08001000 `

closed 2023-03-20 by TheKikGen