TheKikGen/USBMidiKliK4x4 #30: sharing PCB design files for 3x3 shield

kpek, 2021-01-24:
Hello,

First of all thank you very much for this amazing piece of software, the usb/midi devices (which are actually working as expected) are pretty expensive and this is perfect solution for DIY people out there.

I saw you also created a shield PCB for 3x3 interface which you sell on tindie.
Would it be possible to include eagle/kicad design files (or whatever PCB design software you used) in the git repo, maybe also including the BOM ? 

I know you do this for free and the PCB shield you sell on tindie is at least some sort of income, but i think it would really benefit some people, especially from poor countries and countries where the shipping cost would actually be higher than the price for the shield itself ( in which case it would be even cheaper to buy some proprietary interface ).

Thank you !

r3d5ky, 2021-01-25:
But there is a complete schematic in repo, and with minimal knowledge of working with any pcb designing software (Kicad, EasyEDA, etc), you can make your own pcb desing in a couple of hours, and you can make it accordingly to your needs and preferences. It is really good and huge project, and maybe it will be better to spend some time and learn pcb designing a bit than ask creator to give you gerbers and other stuff :)

kpek, 2021-01-25:
Yes, you are right, shortly after i wrote this, i found the schematic and created my own PCB in EasyEDA,  its already being manufactured. Very nice project for first time PCB designers. 
Was just being lazy i guess, with all the other projects iam working on :) anyway this thread can be closed. Thanks.

closed 2021-01-25 by kpek