TheKikGen/USBMidiKliK4x4 #31: cant get to configuration menu in serial mode

kpek, 2021-03-29:
Hi,

Iam trying for multiple days now to get to the configuration menu and cannot get it to show up.
The firmware itself works as expected and whole device performs nicely, but i need to change few things in the configuration and cannot get there.
Iam using bluepill (aliexpress/china clone, yes might be the cause but i ran other simple sketches which print and shows serial output without issues, so i would use that as last resort excuse). Latest version of the bootloader and the firmware.

So what i tried (everything like 20+ times):
Erased / zeroed out the stm32
Reflashed tkg bootloader (from binary, but also compiled my own)
Reflashed the firmware (from binary, but also compiled my own)

After this, i run the command from documentation: sendmidi dev "MidiKlik 4x" syx hex 77 77 78 06 08
Device reboots, and shows as virtual COM device, so far so good.
I try to connect to the COM device using seria software
Theres no output.
Mashing all kinds of buttons and sending that to the device, nothing shows up.

I also tried this on multiple computers (2x win, 1x mac) using different softwares (5 or more different ones)

Thres just no output on the serial window on virtual USB COM port on either of the computers.

Am i doing something wrong or is there an issue hidden somewhere ?

Many thanks in advance.

kpek, 2021-03-30:
one additional test i performed:
i found this simple sketch on the internet, that basically tests the serial communication (sending and receiving) and this works totally fine. Iam still using latest tkg bootloader and it worked on first try, in this case using simply the serial monitor in arduino IDE. Whatever character i enter and hit submit, bluepill then responds with hello and the character i entered.

> _int8_t led = 0;
> 
> void setup() {
>    // Serial.begin(); //this isn't needed, it is pre-initialised , otherwise you could uncomment it to test
>    pinMode(LED_BUILTIN, OUTPUT);
> }
> 
> void loop() {
>   if(Serial.available()) {
>     char c = Serial.read();
>     Serial.print("hello ");
>     Serial.print(c);
>     Serial.println();
>   }
>   led = ~ led & 1;
>   digitalWrite(LED_BUILTIN, led);
>   delay(100);
> }

[image]

TheKikGen, 2021-04-18:
Hello.

What serial terminal software are you using ?
Baud rate must be 115200

kpek, 2021-04-19:
Hello,

As mentioned initially, i tried 5 different terminal software - screen, putty, tera term, minicom and arduino ide terminal.
Always used 115200 baud rate (also tried different baud rates just for test without any difference).
Also used 2 different blue pills, and same issue for both.

I recall some time ago (around 2 years ago) i used Usb MidiKlik on arduino uno (clone), and this serial configuration menu popped up immediately and  worked just fine.

BR,
Adam

TheKikGen, 2021-05-03:
Hi,

I think this issue is linked with your Bluepills.

Edit the source, and try to force the serial mode :  
https://github.com/TheKikGen/USBMidiKliK4x4/blob/fa41e207d4e11e76ece4eabad9b71a2f76f5a108/UsbMidiKliK4x4.ino#L853

Replace the first lines of CheckBootMode() to force the config mode with  : 
``````
// start USB serial
Serial.begin(115200);
delay(1000);

while (!Serial) {
          LED_Flash(&LED_ConnectTick);delay(300);
          LED_Flash(&LED_ConnectTick);delay(300);
          LED_Flash(&LED_ConnectTick);delay(300);
}
digitalWrite(LED_CONNECT, LOW);

ShowConfigMenu(); 

````
Save, compile and upload that test firmware, and retry...

kpek, 2021-05-17:
Hi,

Yes this works, it goes straight into the serial mode and config menu appears immediatelly after booting.

[image]
[image]

I started commenting out some lines in that function to find out what causes this and commenting out following 2 lines solves this problem and it works as it should:

//      Wire.begin();
//      Wire.setClock(B_FREQ) ;

[image]

Not sure if this will cause any other issues though, with those 2 lines commented out ? 
Please let me know, iam not planning on chaining multiple units together, but having only single standalone 3x3 interface.

Thanks alot.
Adam

TheKikGen, 2021-05-19:
ok. Thanks Adam.
You commented I2C Bus feature.  
Maybe you should update the STM32DUINO core sources, as I2C Wire library was updated many times.
I had this kind of issue with faulty Bluepills before.  
At end : you could also use one of the pre-compiled binary provided here : https://github.com/TheKikGen/USBMidiKliK4x4/releases

iflyhigh, 2022-12-27:
Faced exactly the same issue as described above, but it persists both with pre-built binaries and compiled ones using latest 1.8.19 IDE, core and libraries (reinstalled from scratch). Commenting out Wire.begin() helps but it looks like board chaining feature will be lost.

KittenVillage, 2023-06-22:
I'm flashing a few Bluepill boards and have run into this error. Just from reading through the discussion I wondered if it had/has something to do with I2C. So after reading up on the busboard page I ran 2 10k resistors from 5v to PB6 & PB7. Sent the sysex and got into the serial menu no problem. Am still testing, but if you run into this, try resistors! (Will this set a master/slave internal toggle to eprom or something? Will it work after this?)