TheKikGen/USBMidiKliK4x4 #32: Problem in version 2.5

MSMicro-ru, 2021-11-13:
Good work!
In version 2.5 (testing on blackpill F103c8, 64K firmware umk4x4.stm32f103c8-64k-o2.bin) :
after sending command: Reboot in configuration mode (F0 77 77 78 06 08 F7) - 
device is reboot in update mode!
USB\VID_1209&PID_BEBA\
After reflashing - device onсe enter to config mode (COM Port is on),
and after - no entering to configuration mode.

TheKikGen, 2021-11-15:
At the first boot, configuration mode is enabled automatically.
If you save the configuration, the next time, the board will boot as a midi interface.
In  configuraiton mode, try to reset all parameters, save and exit.
Sometimes behaviour can differ between BluePill boards (not the same resistors, STM32 clones, etc...)

MSMicro-ru, 2021-11-15:
Yes,, all works, as you say, but
after sending command: Reboot in configuration mode (F0 77 77 78 06 08 F7) -
device is reboot in UPDATE mode!

closed 2023-03-20 by TheKikGen