TheKikGen/USBMidiKliK4x4 #34: Does the firmware work with the silver colored version?

DepreTux, 2021-12-15:
Do you know if it will work with this version:

https://articulo.mercadolibre.com.ar/MLA-671515228-interfaz-midi-usb-midiplus-midilink-midi-4x4-para-teclado-_JM

It's the only one I could find here

Amazing project, anyway. I'm surprised this is not better known within the electronic music community (at least I just found it).

Thanks for the firmware and the help!

TheKikGen, 2022-01-17:
Help us to make ths project to be better known !
The midi interface looks like exactly the one I hacked.  Midiplus and Miditech are the same company.
So I'm almost sure that the hack will work on it.

jnovek, 2022-07-19:
I'm working on the silver MIDIPLUS box right now and it would be nice if someone could sanity check this before I set it up.  I'm pretty new to STM32 as a platform in general.

From the Miditech instructions (https://github.com/TheKikGen/USBMidiKliK4x4/wiki/The-Miditech-Midiplus-4x4-hack-and-STM32Duino) it seems like you need to:
  - Remove R28
  - Connect boot0 directly to 3.3v
  - Connect boot1 to GND via 10k pulldown
  - Add jumpers to access RX1 and TX1 via serial programmer
  - Add a jumper to GND or RST so I can reset the device
  
Anything I'm missing? I've marked all the spots on the attached photo of the silver MIDIPLUS board.

Questions:
[image]

  1) You mentioned on this page that you had to remove "multiple" resistors, but the only one I see is R28.  Am I missing something?
  2) Is there a better place to get at boot0 than R27?  I suspect not -- it looks like it comes right off of pin 80 -- but that's going to be some dicey soldering for me.  I think I'll have to expose some of the trace to properly get at it.  I could also replace R27, it appears to be 10k to GND.

TheKikGen, 2023-01-03:
Sorry for that very late answer !
I think you could just solder a connector at CN1 and use an ST-LINK V2 Dongle clone to flash directly the firmware.
When I hacked this board, I haven't any ST-LINK dongle available, so I used the internal serial bootloader.

TheKikGen, 2023-01-03:
Check my other project "Kikpad" here for the Midiplus Smartpad : https://github.com/TheKikGen/kikpad

Pinout is :
1.GND
2.SWDIO
3.SWDCLK
4.RESET

closed 2023-12-20 by TheKikGen