TheKikGen/USBMidiKliK4x4 #41: Wiki: Circuit Mod Error

PatrickECleary, 2023-01-06:
Hi y'all 👋🏼

I found an error on the Wiki article "[The Miditech Midiplus 4x4 hack and STM32Duino](https://github.com/TheKikGen/USBMidiKliK4x4/wiki/The-Miditech-Midiplus-4x4-hack-and-STM32Duino)".

The [RST] callout in the [fifth](https://github.com/TheKikGen/USBMidiKliK4x4/wiki/miditech-hack/miditech4-board5.jpg?raw=true) image down is pointing to the [+3.3V] line coming from the LDO, not the [RST] pin.

It looks like R26 is a pull-up resistor. In that image, the top of R26 is [3v3] and the bottom is [RST]. Note that its neighbor, C16, is decoupling [RST] with respect to ground.

Although this wiring may be working for folks, it's bad practice and should be corrected. While I haven't tested it, I'm guessing that the reset pulse sent by the TTL to serial converter may be sufficiently brief to pull the LDO output low enough that the uC resets. However this shorts the LDO through the TTL to serial converter every time. 

Thanks for your work, TheKikGen team! This project writeup is fantastic 🙏🏼

PatrickECleary, 2023-01-06:
_A handy note:_ On my board (V0.25 20200213) the [RST] pin is exposed at **Pin 4** of the unpopulated **CN1** connector right below the USB port. This is far easier to solder to than a tiny SMT pad :)

TheKikGen, 2023-01-11:
Thanks Patrick.
You are right.   RST is at the bottom of R26.

You can also find a RST at the CN1 connector (not soldered) being a JTAG/SWD you can use with a STLINK stick.

closed 2023-12-20 by TheKikGen