TheKikGen/USBMidiKliK4x4 #49: Incorrect check for number of string descriptors

M-HT, 2024-01-15:
In function usb_midi_GetStringDescriptor there's a [check for size of structure](https://github.com/TheKikGen/USBMidiKliK4x4/blob/e507929189d6c45175801a1d964c98eaf09b82bc/usb_midi_device.c#L558) instead of check for number of string descriptors.

I think this line
`if (wValue0 >= sizeof(usbMIDIString_Descriptor) ) {`
should be replaced with
`if (wValue0 >= (sizeof(usbMIDIString_Descriptor) / sizeof(usbMIDIString_Descriptor[0])) ) {`

TheKikGen, 2024-12-31:
Correction done in release 2.53.

TheKikGen, 2024-12-31:
This didn't impact UMK behavious at all at the moment, but code was incorrect;  Thanks to @M-HT !!

https://github.com/TheKikGen/USBMidiKliK4x4/releases/tag/v2.5.3

closed 2024-12-31 by TheKikGen