TheKikGen/USBMidiKliK4x4 #53: Version 2.5.2 doesn't work in standalone mode from usb charger

DanielHabart, 2024-12-03:
Version 2.5.2 doesn't work in standalone mode from usb charger. Gets stuck on all LEDs on. No such problem when connected to the computer but without sending any midi info and hence turning into standalone mode by itself.

TheKikGen, 2024-12-23:
Hi,  did you test any previous version ?  I have the same behaviour with my Midiface also.

DanielHabart, 2024-12-30:
Thanks for getting back to me. I then tried a previous version and that worked without problems. I think it was the version without the new bootloader (?).Maybe it was just an upload issue, although I assume I must have tried uploading multiple times.

TheKikGen, 2024-12-31:
Bug corrected in the 2.53 release, published today.
https://github.com/TheKikGen/USBMidiKliK4x4/releases/tag/v2.5.3

closed 2024-12-31 by TheKikGen